xiaoyaoyou1212/BLE #4: api 23 以上 需要动态申请权限

iYoullloveGitHub, 2017-02-27:
api 23 以上 需要动态申请权限, 否则就无法显示连接的蓝牙设备

xiaoyaoyou1212, 2017-03-04:
@iYoullloveGitHub 目前这个还没做适配，后面会加上

kanshenmekan, 2017-03-21:
加上了动态申请，还是扫描不到设备呢

kanshenmekan, 2017-03-21:
小米的系统扫不到设备，动态申请了权限。。。

xiaoyaoyou1212, 2017-03-21:
@kanshenmekan 我用demo试过了，6.0系统是可以扫描到设备的呀，需要注意的是这里只针对低功耗设备哦

kanshenmekan, 2017-03-21:
startScan里面的回调方法onDeviceFound 里面什么都没有，应该说都没调用到

kanshenmekan, 2017-03-21:
if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.M) {
            if (this.checkSelfPermission(Manifest.permission.ACCESS_COARSE_LOCATION) != PackageManager.PERMISSION_GRANTED) {
             requestPermissions(new String[]{Manifest.permission.ACCESS_COARSE_LOCATION}, 1);
            }
        }
不知道是不是这个有问题吗，小米系统是申请的时候给的权限是 询问，当你要用的时候才会弹框，但是我调用了你的startScan连弹框都没有

xiaoyaoyou1212, 2017-03-21:
@kanshenmekan 这个看不出什么的，没扫到设备就没有该回调，你用其他的手机试过可以扫到那个设备么，我这里没有6.0以上系统的小米手机，所以也看不出你这个问题所在，我这个demo进入应用就会申请蓝牙权限，如果拒绝就会直接退出的

kanshenmekan, 2017-03-21:
@xiaoyaoyou1212 哦哦，谢谢哦，那应该是我权限这边有问题吧，这些厂家都自己做自己的了。。跟谷歌的规则都不一样了。搞得好难啊

kanshenmekan, 2017-03-22:
@xiaoyaoyou1212 老哥能发一下 你6.0版本以上demo的apk给我吗。。我装一下，看看我这系统能搜索到设备吗。。1207769754@qq.com

xiaoyaoyou1212, 2017-03-23:
@kanshenmekan demo已发，GitHub上也已修改，增加了权限适配

kanshenmekan, 2017-03-23:
@xiaoyaoyou1212 装了一下， 和我之前用原生代码写得一样。。我刚开始是以为搜索不到设备，后来搜索到了一个，但是一般的手机设备都搜索不到，你发给我这个apk也是一样。我这边测试机还没过来，可能小米这系统真有问题吧。。讲道理5.0以后的手机都应该能被搜索到。我俩手机都是小米的。。。

xiaoyaoyou1212, 2017-03-23:
@kanshenmekan 这个demo我用乐视手机6.0系统试过了是可以的，只能搜到BLE设备，不能搜到手机哟，至于小米手机扫描不到设备我这里没测试机也不好说这个问题是怎么回事。

kanshenmekan, 2017-03-23:
@xiaoyaoyou1212 手机不算是ble设备吗。。不是说系统在5.0之后都能作为服务端了。可以被startlescan方法扫描到。。而且手机的蓝牙都是低功耗和经典蓝牙都兼容的

xiaoyaoyou1212, 2017-03-23:
@kanshenmekan 手机不是BLE设备额，BLE专属低功耗，手机是双模，可兼容单模的BLE设备，如果概念上有不清晰的可以看看我的博客，里面有一篇讲相关概念的。目前Android5.0以后可支持充当外设被扫到，但是需要使用5.0那个新的扫描方式，我的demo中还没使用该方式，但是蓝牙库已经支持，版本说明中有介绍，你可以使用新的扫描方式试试。

kanshenmekan, 2017-03-23:
@xiaoyaoyou1212 好，谢谢，我刚接触蓝牙这块，还是小白，自己在慢慢研究。

kanshenmekan, 2017-03-23:
@xiaoyaoyou1212 老哥，有问题的时候能请教一下吗，有时候会的人两句话，自己弄得话，可能要搞个几天的。

xiaoyaoyou1212, 2017-03-23:
@kanshenmekan 嗯，你可以加我QQ的，这样聊不是很方便，我的博客中有我的联系方式

xsfgdut, 2017-03-23:
那个在Android 7.0系统上可以扫描到设备但是连接不上，这是怎么回事？

closed 2017-03-24 by xiaoyaoyou1212