xiaoyaoyou1212/BLE #7: 选择Unknown Service的时候会抛出异常.

MAIMIHO, 2017-03-15:
[image]
`03-15 10:24:39.052 9652-9652/com.vise.ble E/AndroidRuntime: FATAL EXCEPTION: main
                                                            Process: com.vise.ble, PID: 9652
                                                            java.lang.ClassCastException: android.bluetooth.BluetoothGattDescriptor cannot be cast to android.bluetooth.BluetoothGattCharacteristic
                                                                at com.vise.ble.DeviceControlActivity$7$1.a(DeviceControlActivity.java:331)
                                                                at com.vise.baseble.a$4$8.run(ViseBluetooth.java:273)
                                                                at android.os.Handler.handleCallback(Handler.java)
                                                                at android.os.Handler.dispatchMessage(Handler.java)
                                                                at android.os.Looper.loop(Looper.java)
                                                                at android.app.ActivityThread.main(ActivityThread.java)
                                                                at java.lang.reflect.Method.invoke(Native Method)
                                                                at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java)
                                                                at com.android.internal.os.ZygoteInit.main(ZygoteInit.java)
03-15 10:28:26.203 2523-3930/? E/InputDispatcher: channel 'fc825a0 com.vise.ble/com.vise.ble.DeviceScanActivity (server)' ~ Channel is unrecoverably broken and will be disposed!
03-15 10:28:26.203 2523-3930/? E/InputDispatcher: channel '4f8f691 com.vise.ble/com.vise.ble.DeviceDetailActivity (server)' ~ Channel is unrecoverably broken and will be disposed!`

xiaoyaoyou1212, 2017-03-15:
@MAIMIHO 问题已改，看看是否还会出现

MAIMIHO, 2017-03-15:
谢谢, 我现在查看一下

MAIMIHO, 2017-03-15:
现在好了, 谢谢诶.

closed 2017-03-15 by xiaoyaoyou1212