xiaoyaoyou1212/BLE #13: 超过20自己的数据，怎么分包发送

Darksiderlyd, 2017-03-23:
最开始的实现方式是循环发送：
for (int i = 0; i < paramStr.length(); i += 20) {
    // 预加 最大包长度，如果依然小于总数据长度，可以取最大包数据大小
    if ((i + 20) < paramStr.length()) {
        String rangeStr = paramStr.substring(i, i + 20);
        Log.d(BLETAG, rangeStr);
        Write(mWriteCharacteristic, rangeStr);
    } else {
        String rangeStr = paramStr.substring(i, paramStr.length());
        Log.d(BLETAG, rangeStr);
        Write(mWriteCharacteristic, rangeStr);
    }
}

第二种是这样：
private void nextRequest(BluetoothGattCharacteristic mCharacteristic) {
        if (mOperationInProgress)
            return;
        // Get the first request from the init queue
        String request = orderItemQueue != null ? orderItemQueue.poll() : null;
        if (request == null) {
            return;
        }
        mOperationInProgress = true;
        boolean result = false;
        result = Write(mCharacteristic, request);
        // The result may be false if given characteristic or descriptor were not found on the device.
        // In that case, proceed with next operation and ignore the one that failed.
       
            mOperationInProgress = false;
            nextRequest(mCharacteristic);
    }

    private void Write2(BluetoothGattCharacteristic mCharacteristic, List<String> paraStrs) {
        orderItemQueue.clear();
        orderItemQueue.addAll(paraStrs);
        nextRequest(mCharacteristic);
}

第一次没用队列那样发送成功路就很高120多个字节 丢包率很小,
第二次硬件测试表示，丢包率很高  

这是写入的时候捕获设备是否初始化的那个方法
handleAfterInitialed(getBluetoothGatt().writeCharacteristic(characteristic), bleCallback);
发送失败的时候（丢包的时候）就会返回这个 bleCallback.onFailure(new InitiatedException(),"");异常
大神我该怎么分包，按道理是不是我发送了就应该没问题了

Darksiderlyd, 2017-03-23:
有没有交流群或者QQ能否加一下，一起讨论下

xiaoyaoyou1212, 2017-03-23:
@Darksiderlyd 需要按照每包20字节进行分包处理，可以使用队列进行处理，记得每包之间发送需要一定时间的间隔，一般一两百毫秒

xiaoyaoyou1212, 2017-03-23:
@Darksiderlyd 我QQ553252628，其实博客中有联系方式的，最近很多问分包相关的，我后面把分包的相关处理方式整理下发出来

Darksiderlyd, 2017-03-23:
蓝牙那边 接收时间 只有一秒 一秒把我发过去的150个字节 或者是120 个字节接收到  隔200ms就超时了 蓝牙那边不让做延时，他说要确认上一包数据发送完成之后在发送下一包  我也是醉了 怎么确认上一包数据发送完成。。。。

Darksiderlyd, 2017-03-23:
硬件不是我们公司的是别的公司的交流起来很难

xiaoyaoyou1212, 2017-03-24:
@Darksiderlyd 150字节一秒是能接收完的，但是这个分包是必须的，你把150字节分成8个包，每包发送间隔用100毫秒这样试试。

Darksiderlyd, 2017-03-24:
嗯嗯我试一下

Darksiderlyd, 2017-03-28:
谢谢大神  我看到了那个分包的了 我搞一下打个包给硬件那边

closed 2017-03-28 by xiaoyaoyou1212

keywish, 2019-01-11:
我看库是由 分包发送 api，但是调用没有起作用，后面要是库直接支持就i好了