xiaoyaoyou1212/BLE #40: 会扫出大量重复的设备，希望作者解决

ervinsae, 2017-11-27:


xiaoyaoyou1212, 2017-11-27:
你确定设备有重复？下面的mac地址可以区分设备唯一性的，不要依据设备名称。

ervinsae, 2017-11-27:
扫描回调里面打印出来的，就是循环打印这些设备

11-27 15:04:37.631 8190-8190/ D/DEVICE: 58:E2:88:4C:A7:6D
11-27 15:04:37.631 8190-8190/ D/DEVICE: D9:08:45:C4:F1:66
11-27 15:04:37.661 8190-8190/ D/DEVICE: 55:84:DC:F6:AD:DB
11-27 15:04:37.721 8190-8190/ D/DEVICE: 66:B9:7E:EC:09:BE
11-27 15:04:37.731 8190-8190/ D/DEVICE: D9:08:45:C4:F1:66
11-27 15:04:37.781 8190-8190/ D/DEVICE: D9:08:45:C4:F1:66
11-27 15:04:37.821 8190-8190/ D/DEVICE: 58:E2:88:4C:A7:6D
11-27 15:04:37.821 8190-8190/ D/DEVICE: 58:E2:88:4C:A7:6D
11-27 15:04:37.821 8190-8190/ D/DEVICE: D9:08:45:C4:F1:66
11-27 15:04:37.861 8190-8190/ D/DEVICE: D9:08:45:C4:F1:66
11-27 15:04:37.881 8190-8190/ D/DEVICE: 2D:0E:D4:6C:65:DC
11-27 15:04:37.901 8190-8190/ D/DEVICE: 71:B3:01:01:99:D9
11-27 15:04:37.901 8190-8190/ D/DEVICE: D9:08:45:C4:F1:66
11-27 15:04:37.911 8190-8190/ D/DEVICE: 66:B9:7E:EC:09:BE
11-27 15:04:37.931 8190-8190/D/DEVICE: 55:84:DC:F6:AD:DB
11-27 15:04:37.951 8190-8190/ D/DEVICE: 04:52:C7:DC:E7:4D
11-27 15:04:37.951 8190-8190/ D/DEVICE: D9:08:45:C4:F1:66
11-27 15:04:37.981 8190-8190/ D/DEVICE: 2D:0E:D4:6C:65:DC
11-27 15:04:37.981 8190-8190/ D/DEVICE: D9:08:45:C4:F1:66
11-27 15:04:38.001 8190-8190/ D/DEVICE: 58:E2:88:4C:A7:6D
11-27 15:04:38.001 8190-8190/ D/DEVICE: 58:E2:88:4C:A7:6D
11-27 15:04:38.031 8190-8190/ D/DEVICE: D9:08:45:C4:F1:66
11-27 15:04:38.091 8190-8190/ D/DEVICE: 2D:0E:D4:6C:65:DC

xiaoyaoyou1212, 2017-11-27:
这个是因为设备信号值有更新，界面上只是更新不会增加设备的

closed 2017-12-12 by xiaoyaoyou1212