xiaoyaoyou1212/BLE #47: 连接失败多次问题

Jys05, 2018-03-19:
当连接过程中，多次连接失败，会导致app卡死，以及报线程过多

Jys05, 2018-03-19:
[connect.txt](https://github.com/xiaoyaoyou1212/BLE/files/1823639/connect.txt)
[image]

xiaoyaoyou1212, 2018-03-20:
@Jys05 好的，谢谢建议，我抽时间优化下这个

Jys05, 2018-03-21:
是的，我看到您的方法加入了synchronized，应该没有问题，但是在公司项目中由于设备蓝牙设备问题，会连接多次失败，然后就发现app会卡顿！

Jys05, 2018-03-21:
这些修改的代码，我是从另一个库移植过来的！使用您的库的demo连接公司的设备时也会卡顿，但是使用另一个库的没有这个问题，由于您写的写入和通知等方法太优秀，所以没有使用那个库！我找一下，我再贴上来，请您看看

Jys05, 2018-03-21:
我当时修改后多次连接失败就不卡顿了，也不报错。
您看看，以下是我参考的library以及nrf connect软件：
1、[https://github.com/NordicSemiconductor/Android-BLE-Library/blob/master/ble/src/main/java/no/nordicsemi/android/ble/BleManager.java](url)
2、[http://www.wandoujia.com/apps/no.nordicsemi.android.mcp](url)

[image]

Jys05, 2018-03-21:
我的处理是：当连接失败，就重新扫描

xiaoyaoyou1212, 2018-03-21:
@Jys05 好，我看能否优化下这个，但有一个问题没明白，你这个有多线程操作需要么，按道理一般的单线程可以满足需求的。还有上面写的延时200ms会导致界面卡顿的，貌似这样的解决方案不是太好。

Jys05, 2018-03-23:
@xiaoyaoyou1212 您好，重新描述一下情况是这样的多次失败之后，再次扫描会报：“Skipped 120 frames!  The application may be doing too much work on its main thread.”，当时开发一直以为自己写的app内存泄漏导致，最后的解决是将您的库以module形式加入并修改上述代码位置。，可能上述的解决方式不太好，您看看！

Jys05, 2018-03-23:
使用Ble软件的LogCat：
测试设备：是公司的打印机
[bleDemo_Logcat.txt](https://github.com/xiaoyaoyou1212/BLE/files/1839928/bleDemo_Logcat.txt)

xiaoyaoyou1212, 2018-03-23:
@Jys05 你有加QQ群没，可以进群交流下，这样沟通不是太方便。看日志是由于在连接失败后频繁点击连接操作导致的，你这样修改后确实能避免重复连接，但是我的建议是上层做下防重复点击功能。

Jys05, 2018-03-23:
@xiaoyaoyou1212 好的

xiaoyaoyou1212, 2018-04-26:
@hjs-wcl 没太明白没有连接成功调用disconnect的作用是？不能这样使用的

hjs-wcl, 2018-04-27:
我的理解是如果调用了connectGatt，在指定的时间内，如果没有连上设备，再去重连，如果连接成功了，会导致多次调用callback的问题。

SJLBenjamin, 2019-12-05:
> work

我这边也出现这样的问题,一直连接失败,我的操作是连接失败就重新搜索连接,也会出现这样的Skipped 120 frames! The application may be doing too much work on its main thread.提示 不知道你是怎样解决的