xiaoyaoyou1212/BLE #66: 从1.0升级到2.0，数据发送不了

lanmanck, 2018-12-14:
老兄，之前用的是你的1.x版本没啥问题，就是我之前说的8.0版本问题，现在我在升级到2.0时，用demo apk都发送不了，写的service，但是提示是read啊，这可怎么搞？你这API变化也太大了。
if (bleCallback != null && bluetoothGattChannel != null) {
            String key = bluetoothGattChannel.getGattInfoKey();
            PropertyType propertyType = bluetoothGattChannel.getPropertyType();
            if (!bleCallbackMap.containsKey(key)) {
                bleCallbackMap.put(key, bleCallback);
            }
            if (propertyType == PropertyType.PROPERTY_READ) {
                if (!readInfoMap.containsKey(key)) {
                    readInfoMap.put(key, bluetoothGattChannel);
                }
            } else if (propertyType == PropertyType.PROPERTY_WRITE) {
                if (!writeInfoMap.containsKey(key)) {
                    writeInfoMap.put(key, bluetoothGattChannel);
                }
            } else if (propertyType == PropertyType.PROPERTY_NOTIFY) {
                if (!enableInfoMap.containsKey(key)) {
                    enableInfoMap.put(key, bluetoothGattChannel);
                }
            } else if (propertyType == PropertyType.PROPERTY_INDICATE) {
                if (!enableInfoMap.containsKey(key)) {
                    enableInfoMap.put(key, bluetoothGattChannel);
                }
            }
        }
[image]
[image]
[image]
在 showGattServices()的时候，为什么要
BluetoothDeviceManager.getInstance().read(mDevice);
一下呢？
我这个2个if都进去了，先 read，然后再setnotify()
BluetoothDeviceManager.getInstance().bindChannel(mDevice, PropertyType.PROPERTY_NOTIFY, service.getUuid(), characteristic.getUuid(), null);
                    BluetoothDeviceManager.getInstance().registerNotify(mDevice, false);

lanmanck, 2018-12-14:
我这个service可以read write_no_response notify.现在写不出去，writeInfoMap为NULL

xiaoyaoyou1212, 2018-12-15:
2.x修改是有点大，但是基础的功能还是没变，demo apk为了兼容所以会有一定的判定，如果你这个UUID没有写入属性在demo中是无法写入的。你可以写死UUID在demo中试验下，把写入的UUID就用提供的那个UUID，然后发送数据再试试~